raul-ortega/u360gts #4: Inverted servos support

raul-ortega, 2016-11-02:


jelle737, 2017-09-19:
Just in case this would be a problem for someone there is a mechanical solution for this. Inside the 360° servo switch the two motor wires going to the brushed motor. This will make 360° servo move in the other direction with the existing PID controller and in turn give the correct behaviour.

jelle737, 2018-06-07:
Inverting the behaviour of the pan servo, it seems that in my case I have to inverted it (and I didn't want to resolder the motor wires as this upgraded 360° servo is much fancier, can be accomplished in the following fashion by inverting the value written towards the servo.
In the following file:
https://github.com/raul-ortega/u360gts/blob/f6f41eefc770b2c448f630c071d058becc1af0e3/src/main/tracker/servos.h#L35-L37
Thus change:
`pwmWriteServo(panServo, A);`
To: This should make sense as the pan0 and all other calculations stay true to their logic, only the output is totally different.
`pwmWriteServo(3000 - panServo, A);`

However if a person know that 1000µS turns counter-clockwise, and sees it running clockwise he might suspect something odd like this. But for everyone telling that the servo restingpoint pan0 is 1300 (where it would be servo-tester-tested 1700) actually is quite ok, because I think everyone will just use the in program code. Although I do suspect the autocalibrate function does not work with this quick mod.

raul-ortega, 2019-09-11:
I have uploaded an experimental version (dev-10.3.0) which would allow to use inverted pan servos.

The user need to configure it as follows from CLI tab:

set pan_inverted=on

save

No calibration modifications have been introduced in the source code related with inversion, thus the automatic pan servo calibration might be performed in an non desirable way. In that case the user need to do it manually.

I haven't tested it yet, any feedback will be wellcome.

[https://github.com/raul-ortega/u360gts/blob/master/other-releases/amv-open360tracker_dev10.3.zip](https://github.com/raul-ortega/u360gts/blob/master/other-releases/amv-open360tracker_dev10.3.zip)

raul-ortega, 2020-09-25:
Solved since release [11.0.0](https://github.com/raul-ortega/u360gts/releases/tag/11.0.0)

closed 2020-09-25 by raul-ortega

ikwidtech, 2020-11-26:
Im trying this build this. Do you have any more detailed info on building the servos and the gearing to spin it?

raul-ortega, 2020-11-26:
This project does not cover building servos but buying 360º continuous rotation servos or converting 180º degrees servos to 360º continuous rotation (searching google will give you tons of tutorials about conversion conversion).

ikwidtech, 2020-12-08:
Thanks im just waiting for my servos to come in the mail as well as the
screen