raul-ortega/u360gts #17: Changing magnetometer alignment when using external compass

jelle737, 2017-09-27:
When using a compass with magnetometer together with a naze32 acro version. Usually this chip is at the bottom of the board, this gives counter clockwise readings. The align_mag option to configure the magnetometer alignment should be brought back to the cli.
The formatting in cli is as follows:
`set align_mag = CW0FLIP`
The solution to activate this cli line is to uncomment one line in the follow file
https://github.com/raul-ortega/u360gts/blob/b4e3b760e2748d367b8ddc0f25416860ae6f986a/src/main/io/serial_cli.c#L566
Thus change:
`    { "align_acc",                  VAR_UINT8  | MASTER_VALUE | MODE_LOOKUP,  &masterConfig.sensorAlignmentConfig.acc_align, .config.lookup = { TABLE_ALIGNMENT } },
    { "align_mag",                  VAR_UINT8  | MASTER_VALUE | MODE_LOOKUP,  &masterConfig.sensorAlignmentConfig.mag_align, .config.lookup = { TABLE_ALIGNMENT } },
`
to:
`    { "align_acc",                  VAR_UINT8  | MASTER_VALUE | MODE_LOOKUP,  &masterConfig.sensorAlignmentConfig.acc_align, .config.lookup = { TABLE_ALIGNMENT } },*/
    { "align_mag",                  VAR_UINT8  | MASTER_VALUE | MODE_LOOKUP,  &masterConfig.sensorAlignmentConfig.mag_align, .config.lookup = { TABLE_ALIGNMENT } },
/*`
I tried this out and it is working

raul-ortega, 2017-10-03:
I'm still in the process of moving to my home country. I'll update the repository as soon as possible. Thanks so much for your contribution.

raul-ortega, 2018-05-04:
Incorporated to development version

closed 2018-05-04 by raul-ortega