raul-ortega/u360gts #23: GPS Status information.

geofrancis, 2018-05-01:
I am currently using your ground station with a Here+ RTK GPS system on Arduplane wing.

the ground RTK receiver is connected to mission planner over USB and does the rtk packet injection over mavlink to the plane. 

But it also has a standard uart on the receiver so its also connected to the ground station Naze to be used for its home point.

Its all working but there are a couple of small things that would make it much better for rtk use.

1, would be HDOP of ground gps receiver rather than just sat count shown on the gps screen of the oled
2 would be the gps lock type shown on the mavlink screen, either just a number 0-6 or nogps,2d,3d,dgps,float,fixed

raul-ortega, 2018-05-04:
Let's try...

raul-ortega, 2018-05-04:
Yes, could be!

About HDOP, it will be shown in the same line than Lat/Lon.
About fix type, yes it is possible. Let me think where to put it.

closed 2018-10-08 by raul-ortega