raul-ortega/u360gts #26: RTL/Loiter

tsvetanminchev, 2018-06-19:
I use Farsky- X Telemetry through the BT module from Taranis. When I switch aircraft modes / Stab, RTL, Loiter / tracker turns and stops running. What could be the reason ???

raul-ortega, 2018-10-28:
It might be related with your flight controller firmware, your taranis, your TX/RX system, or your bluetooth modules. Which flight control software are you using?

tsvetanminchev, 2018-12-04:
FK is pixhawk , apm 3.9 plane, frsky  8 r, teensy 3.1 with MAVLink to FrSky S.Port Protocol Converter , BT module 6 master slave. Tranis  x9d+ and Lua telemetry script.

All this work fine  in tracker but I have 20 degree  horizontal lag. When I switch aircraft modes / Stab, RTL, Loiter / tracker turns and stops running.

raul-ortega, 2019-05-22:
The laga of 20 degrees might be caused by many factors:

- offset_trim set to a value between -20 and 20 (<of: > field on screen). Tray changing it from CLI, or by pressing home and menu buttons when tracking is started.

- If magnetic declination param is set to 0 it also might add some degrees of lag.

- NOPID contro enabled add a big lag.

Take a look to this [video](https://www.youtube.com/watch?v=5ddPaFRDjLM)

It is a video demostration published by Carlos Fernández, comparing NOPID and PID controls on pan servo. He built his own customized tracker, and visited me and tested it toguether with me.

In the first part of the video, NOPID control feature is enabled. In the second one NOPID is disabled and PID control has been properly configured for the test. You may judge by yourself which control does the best tracking.

Both tests are done with same flying wing, telemetry protocol, and frequency:

- Reptile 800 SkyShadow
- FrSky SmartPort
- 4 Hz

About your problem with the BT module and tracker stopping, It is quite difficult to determine the origin of the problem. Could you try to capture telemetry in both extremes of the comunication channel at the same time? This would be helpful in order to try to get some data.

raul-ortega, 2019-05-22:
I forgot to tell you that this tracker supports Passthrought FrSky Protocol from Ardupilot. I'm using it in a airplane with pixhack 2.8.4 and R9MM receiver. The pixhack provides telemetry  to the R9MM module through a diode. Inside my Taranis X9D+ I installed  a bluetooth  + signal inverted + bec in order`in order to send the telemetry to the tracker. I also use lua telemetry... I'm using it with success and do not experience any extrange behavior.

In order to use this protocol in the configurator you can activate the AUTODETECT feature or select the FRSKY X protocol in u360gts-configurator and, in addition, you have to set telemetry provider as APM10

Take a look to these links:

[http://ardupilot.org/copter/docs/common-frsky-passthrough.html](http://ardupilot.org/copter/docs/common-frsky-passthrough.html)

[https://github.com/raul-ortega/u360gts/blob/master/docs/SmartPort-Telemetry-from-Taranis-X9D-Plus.pdf](https://github.com/raul-ortega/u360gts/blob/master/docs/SmartPort-Telemetry-from-Taranis-X9D-Plus.pdf)

closed 2019-05-22 by raul-ortega