raul-ortega/u360gts #27: Master does not build for SPRACINGF3

Landflieger, 2018-07-29:
master does'nt build for SPRACINGF3

```
/tmp/cclH0Ray.ltrans8.ltrans.o: In function `calculate_Gtune':
/home/pan/development/u360gts/./src/main/flight/gtune.c:195: undefined reference to `blackboxLogEvent'
collect2: error: ld returned 1 exit status
Makefile:701: recipe for target 'obj/main/amv-open360tracker_SPRACINGF3.elf' failed

```

Landflieger, 2018-07-30:
it does build if I disable blackbox in target.h

raul-ortega, 2018-07-30:
Try compiling development version, I commented those lines there.

You are welcome to be a contributor by forking the repository and sending pull requests. I don't own a spracingf3 board and therefore I have no way to test the firmware on it.

Thanks for your feedback.

raul-ortega, 2018-07-30:
I compiled development version with success. You may download (and unzip) the hex file from here:

[https://www.dropbox.com/s/fzlehuqhq2pdw18/amv-open360tracker_SPRACINGF3.zip?dl=0](https://www.dropbox.com/s/fzlehuqhq2pdw18/amv-open360tracker_SPRACINGF3.zip?dl=0)

Landflieger, 2018-07-30:
hi raul thx ... with commenting the two lines I can build on my own and I flashed successful...
for now I have OLED working, CLI working, can move servos while in CLI mode.
Display setup working partly, can get into menu, but couldn't choose an entry, cause did not find pins for second button ... local GPS isn't working also ... didn't know how to manage uarts ...

so I will have to dive deeper into the code ... but this has to wait for two weeks .. going on holiday tomorrow ... some flying with a friend in Norway ... stay tuned I will continue when I#m back ... If I will it have running completly maybe I can attach some pages for installing with SPRACINGf3 ...

raul-ortega, 2018-07-30:
Thanks again for your feedback and enjoy your holiday!!!

Landflieger, 2018-07-31:
> About buttons:
>
> When you get back on vacation, try connecting the buttons to SONAR pins of IO_2 connector.
>
>  <https://user-images.githubusercontent.com/12935392/43429003-6ca293a0-9461-11e8-9d22-d8392a7d8e42.png>

Hi Raul,

Thank you !! I was wondering if I need an extern pulldown resistor für the buttons … at the moment I#m using normal buttons from a project with Arduino ….