raul-ortega/u360gts #42: The Telemetry input is probably badly initialized

RomanFoltyn, 2019-05-21:
Hallo Raul,
I built it based on your u360gts information. NAZE V10.1.0 efc9492
I am currently testing Hardware and I have the following problem:
When I have a Telemetry signal connected with a BT modem, I can not enter the CLI. It is only the disconnection of connection from the BT signal that enables work via the CLI.
It did not give me peace and I did the following measurements.
The Telemetry Input is constantly 3.3V
The input should have a high impedance, but this input gives a strong output current of around 26 mA. I thought I damaged the processor but its replacement did not help. the effect I have the same. This problem probably affects the operation of Buton and even on the GPS Input.

Connecting the signal from the BT input it works because the BT module has enough power to make zero logic despite the high current still around 1.2V. I think the problem lies in the initialization of the processor input. Please check if there is a bug in this routine.

Best Regards 
Roman Foltyn

raul-ortega, 2019-05-21:
Are you using uart1 and usb at the same time? This is not allowed due to a pin of the processor shared between both.

closed 2019-05-22 by raul-ortega

RomanFoltyn, 2019-05-22:
Hallo Raul,
Thank you very much for the information, you are completely right!
Sorry for the confusion.
Greetings from Bavaria
Roman Foltyn
PS See my projects (stabilizers for cameras) at www.FoMa-systems.com
at the bottom of my page is a link to facebook where you will find really interesting information about my stabilizers for cameras

RomanFoltyn, 2019-08-12:
Hallo Raul,

I hope this mail will reach you.

I am sending the files you need for analysis.

Give me a link describing CLI commands. I would like to set PID parameters but I do not know which belongs to PAN and which to Tilt.

Especially the PAN reacts too slowly and unstably.

If you have any working configuration, please send me.

it will be easier for me to adjust your configuration than to start over. 

 

 

 

I read the CRSF signal directly from the NASE telemetry input

 

 

Best regards 

Roman Foltyn 

version
# u360gts / NAZE  dev10.2.0 Aug 11 2019 / 11:13:23 (c517bc2)
# dump

# dump configuration

# feature
feature -VBAT
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -DISPLAY
feature -BLACKBOX
feature -EASING
feature -NOPID
feature -DEBUG
feature -EPS
feature -RSSI_ADC
feature -AUTODETECT
feature VBAT
feature SERVO_TILT
feature GPS
feature TELEMETRY
feature DISPLAY
feature EASING
feature NOPID
feature DEBUG
feature EPS

# parameters
set servo_pwm_rate = 50
set gps_baud = 4
set gps_provider = NMEA
set gps_sbas_mode = AUTO
set gps_auto_config = ON
set gps_auto_baud = OFF
set gps_min_sats = 6
set gps_home_beeper = ON
set p = 2500
set i = 20
set d = 250
set max_pid_error = 10
set max_pid_accumulator = 5000
set max_pid_gain = 500
set pid_divider = 15
set pan_pin = 0
set pan0 = 1500
set pan0_calibrated = 1
set pan_calibration_pulse = 1500
set min_pan_speed = 0
set offset = 90
set offset_trim = 0
set init_servos = 1
set tilt_pin = 1
set tilt0 = 1450
set tilt90 = 2090
set tilt_max_angle = 0
set easing = 2
set easing_steps = 60
set easing_min_angle = 4
set easing_millis = 15
set telemetry_baud = 8
set telemetry_protocol = 1024
set telemetry_min_sats = 6
set telemetry_provider = NONE
set telemetry_home = DEFAULT
set start_tracking_distance = 10
set start_tracking_altitude = 4
set mag_declination = 0
set mag_calibrated = 0
set nopid_min_delta =  0.200
set nopid_map_angle = 90
set nopid_max_speed = 200
set eps = 1
set eps_distance_gain = 100
set eps_heading_gain = 50
set eps_speed_gain = 60
set eps_frequency = 250
set eps_interpolation = OFF
set eps_interpolation_points = 3
set update_home_by_local_gps = OFF
set max_speed_filter = 0
set telemetry_altitude_priority = BARO