raul-ortega/u360gts #43: Autodetect protocol option kills Bluepill

MFlghts, 2019-05-22:
I think I found a bug. When Autodetect protocol option is set for Bluepill FW 10.1 it's impossible to connect to it any more. Other settings are working normal.

raul-ortega, 2019-05-23:
Please share with me your configuration.

MFlghts, 2019-05-23:
Raul, I can reproduce it with two steps:
1. flashing FW with full chip erase.
2. setting Autodetect protocol in configurator to active. Save.

That's it - no more connection can be established. 
Changing any other settings and saving works fine for me.
Alex.

raul-ortega, 2019-05-23:
I have flashed dev10.1 and proceed to follow your steps, and this is the result. Take a look to this [video](https://youtu.be/sOuWd-ksxjg).

edsellar, 2019-05-23:
I just watched the video, i have some bluepills laying around. I can give it a try as well.

What did you use to compile the new GUI?

MFlghts, 2019-05-23:
Then looks like this problem is with my Bluepill. Sorry. 
I have only one, I can't try another.
Here is my video (and it happens with any v10 FW):
https://vimeo.com/338050069

raul-ortega, 2019-05-23:
@edsellar, new GUI is not yet ready, many changes are still pending. You
should use configurator 3.0.

raul-ortega, 2019-05-23:
@Alex, after flashing you disconnect the usb-tll adapter, change the boot
jumper to its 0 position, and coonect again usb-ttl... You also have to
close configurator window and run it again in order to get COM ports
identifiers again, because it does not detect the ports dynamicaly.

With new GUI this problem is solved.

MFlghts, 2019-05-23:
Raul, but you can see that I've checked boxes Calibrated for Compass and Pan, and both settings got saved and opened correctly. 
I didn't disconnect ttl adapter, I only disconnected power wire from Bluepill. 

I have repeated this issue in different orders and setting - the result is 100% the same.

raul-ortega, 2019-05-23:
It might seem a bit offensive to tell this, but try a different usb cable.
That behavior is too extrange...

MFlghts, 2019-05-23:
Not at all! ) I know that USB cable can be a reason for some problems.
In my case I can change any other settings hundred times, and everything works just fine. I can't see how cable can affect this situation but I will try even with other TTL converter. But this bug looks now to be very specific to my Bluepill only.

raul-ortega, 2019-05-24:
I forgot to comment that you do no have MAG and/or OLED connected to I2C.
This causes the firmware to wait for a timeout period, until it react and
continue the starting up proccess.

MFlghts, 2019-05-24:
I did try different cable and TTL converter - same result. I need to buy different BluePill to try.
If no one else has this problem, let's leave it alone, I think.

MFlghts, 2019-08-22:
I've got a new BluePill, and it has the same problem with latest FW.

raul-ortega, 2019-08-23:
Have you tried with latest cinfigurator too?

raul-ortega, 2019-08-23:
One problem that I detected some time ago with configurator is that when I do a short press on configure button some times it does not receive response from the board, so I had to try again with a long press. But this problem affects to all boards. Latest configurator (v4.0) should solve this.

MFlghts, 2019-08-24:
Yes I used latest configurator and 3.0 as well.
I've also noticed that for both Bluepills STM Flasher shows 64kb memory, but it flashes successfully with memory verifying. My boards have PC13 sign next to the led.  I've learned that there might be different revisions also.

With old configurator I also noticed that sometimes it didn't connect to the board after SAVE command, if I press connect too soon. I thing it happens because board initialization might not be finished yet.

Can you please try Autodetect problem with bare Bluepill (without anything connected except of UART-TTL)?
May be it happens when there is no compass and screen are connected.
This is very strange problem, but in this case I have all setting by default, so this definitely excludes that something is messed up with the settings.

Thank you very much!

raul-ortega, 2019-08-24:
Have you tried with a compass connected?

raul-ortega, 2019-08-24:
About connecting after saving, yes, you have to wait at least 5 seconds.

MFlghts, 2019-08-24:
I don't have a spare compass since it's difficult to buy it now. I have to wait for a new version with QMC support then.

closed 2020-02-23 by raul-ortega