raul-ortega/u360gts #44: No Telemetry out for Sprf3 at Softserial

MFlghts, 2019-05-22:
According to my test with latest fw for sp3 I could only get LTM at SS (port 30) at IO1 .
Mavlink and Nmea did'n work when I just selected them from drop down menu.

raul-ortega, 2019-05-26:
I have tested SoftSerial30 port at IO1 and it seems to work me. Please, take a look to this [video](https://youtu.be/GBaUi2DY3nA).

I will give a try also for SoftSerial31 port...

raul-ortega, 2019-05-26:
Softserial31 port at IO2 also works, as you can see in this second [video](https://www.youtube.com/watch?v=V7xL-B8dKrg)

MFlghts, 2019-05-26:
Can you please try Ardupilot Mission Planner instead of Hercules?
The idea of telemetry relay for me was to connect Mission Planner to see plane location in real time. This didn't work for me. Looks like Mission Planner doesn't accept the telemetry relay Mavlink.

raul-ortega, 2019-05-26:
It is something that has been tested in firmware version 10.0:

[https://www.youtube.com/watch?v=I3JlI0O0Kh8](https://www.youtube.com/watch?v=I3JlI0O0Kh8)

raul-ortega, 2019-05-26:
I have tested Mavlink just now, and the result is shown in [this video](https://www.youtube.com/watch?v=QD21ukuojdM).

Plase, be sure you have checked the two calibrated checkboxes (one for mag and thi other for pan servo).

Also, take into account to set home as auto from telemetry, and increas distance and altitude in order to start tracking. Outgoing telemetry starts sending data once tracking has started.

closed 2019-05-27 by raul-ortega

MFlghts, 2019-05-29:
It's hard to believe, but this is second time when everything goes completely wrong for me.
Can you please have a look.
I did all you mentioned above and I was increasing Alt - it didn't change anything.
Something is starting to transfer to SS when I select LTM, but Mavlink sends nothing.
https://vimeo.com/user1690310/review/339204393/aafdbf6e9d

`
# version
# u360gts / SPRACINGF3 10.1.0 May 19 2019 / 12:59:19 (e0a2069)

# serial
serial 0 1 57600 57600 0 115200
serial 2 0 57600 57600 57600 57600
serial 2 0 115200 57600 0 115200
serial 30 512 57600 57600 57600 57600
serial 31 0 57600 57600 57600 57600

# feature
feature -VBAT
feature SERVO_TILT
feature SOFTSERIAL
feature -GPS
feature TELEMETRY
feature DISPLAY
feature EASING
feature NOPID
feature -DEBUG
feature -EPS
feature AUTODETECT

# set
set looptime=1000
set emf_avoidance=OFF
set rssi_channel=0
set rssi_scale=30
set rssi_zoom=35
set servo_pwm_rate=50
set gps_baud=7
set gps_provider=NMEA
set gps_sbas_mode=AUTO
set gps_auto_config=ON
set gps_auto_baud=OFF
set gps_min_sats=6
set gps_home_beeper=ON
set telemetry_inversion=OFF
set battery_capacity=0
set vbat_scale=110
set vbat_max_cell_voltage=43
set vbat_min_cell_voltage=33
set vbat_warning_cell_voltage=35
set align_gyro=DEFAULT
set align_mag=DEFAULT
set align_board_roll=0
set align_board_pitch=90
set align_board_yaw=90
set gyro_lpf=42
set magzero_x=20
set magzero_y=-82
set magzero_z=-374
set p=2500
set i=20
set d=250
set max_pid_error=10
set max_pid_accumulator=5000
set max_pid_gain=500
set pid_divider=15
set pan_pin=0
set pan0=1487
set pan0_calibrated=1
set pan_calibration_pulse=1400
set min_pan_speed=55
set offset=-13
set offset_trim=0
set init_servos=0
set tilt_pin=1
set tilt0=500
set tilt90=1400
set tilt_max_angle=0
set easing=2
set easing_steps=60
set easing_min_angle=4
set easing_millis=15
set telemetry_baud=7
set telemetry_protocol=8
set telemetry_min_sats=6
set telemetry_provider=NONE
set telemetry_home=AUTO
set start_tracking_distance=6
set start_tracking_altitude=0
set mag_declination=1144
set mag_calibrated=1
set min_logic_level=60
set nopid_min_delta= 0.200
set nopid_map_angle=90
set nopid_max_speed=200
set eps=1
set eps_distance_gain=100
set eps_heading_gain=50
set eps_speed_gain=60
set eps_frequency=250
set eps_interpolation=OFF
set eps_interpolation_points=3
set update_home_by_local_gps=OFF
set max_speed_filter=0
set telemetry_altitude_priority=BARO
`

GOESnick, 2019-06-10:
same problem with mine, dont get any relay data from Uart 2 as well the soft serial