raul-ortega/u360gts #45: MFD to LTM telemetry relay doesn't work

MFlghts, 2019-05-24:
There is fully working setup:
u360gts Simulator set to Mavlink 57600 > Bluepill over ttl > SS30 over BT 57600 > Inav Mission Planner. It shows correct movement on the map.

Issue:
If I ONLY change Mavlink to MFD (both in Settings and Simulator) - Inav Mission Planner resets position to zero and nothing else happens.

raul-ortega, 2019-08-07:
This is not a bug, start tracking distance MAY NOT BE SET TO 0.
Please keep this issue closed.

closed 2019-08-07 by raul-ortega

MFlghts, 2019-08-07:
Why do you keep mentioning the distance, while the problem was with ALT? This is different param.
It also good to have it working with 0 Alt if I want to test tracker with a plane in my hands.

MFlghts, 2019-08-07:
Not exactly! 
start tracking distance **MAY BE SET TO 0**
Right now I have setup with both start tracking distance, and start tracking altitude set to 0 and everything works - tracking and telemetry relay too!
BUT - it only works with LTM out, not with Mavlink.
Whatever bug this is or not, such a behavior is not intuitive and convenient.