raul-ortega/u360gts #47: Crossfire native telemetry support, CRSF relay to u360gts tracker

Saxin, 2019-06-18:
**1. Backgorund**
Crossfire by TBS is advanced UHF Lora modulation based RC link + telemetry system operating on 915 / 868 Mhz bands.
2 variants of TX currently available - full TX (2W capable) and micro TX (250mW), while the larger crossfire TX (in it's full version) having builtin Bluetooth module and is able to relay out Mavlink telemetry - today many users report disconnects and unstable operation of the internall BT module hence rendering this option a no go for relaying to antenna tracker.

Crossfire system RX connection to FC dictates the telemetry method used - 
Method 1, considered preferred is native CRSF protocol, the RX is connected to FC via. signle UART and the later is utilized for both RC link and telemetry, this method is supported by the following FC FW: Kiss / Cleanflight / Inav / Betaflight / dRonin and maybe more.... (connection to RX requires x4 cables - RX / TX / VCC / GND).
CRSF protocol specification file - 
[CRSF Rev07 (1).pdf](https://github.com/raul-ortega/u360gts/files/3303256/CRSF.Rev07.1.pdf)

Method 2, crossfire RX is connected and configured to use x2 uarts - one is used for SBUS and a second one for Mavlink protocol transport, this method is mainly supported on Ardupilot FW and Inav to some extent, I've never used this method and I don't know how the telemetry is presented to OpenTX.

**2. OpenTX**
OpenTX supported for a while S. Port mirror option and recently added support for outputting the native CRSF stream "as is". (IMO the name "S. Port Mirror" after this change is confusing.
Battery compartment serial port is used: serial RX --> Inverter --> BT module.
Related issue: https://github.com/opentx/opentx/issues/5357
Related PR: https://github.com/opentx/opentx/pull/6326
Baudrate is set to 115200 in openTX and BT module accordingly.

**3. CRSF implementation examples**
a. RavenLRS (made by few Inav devs) https://github.com/RavenLRS/raven , 
Relevant files: crsf.c / crsf.h 
https://github.com/RavenLRS/raven/tree/master/main/protocols

b. OpenTX https://github.com/opentx/opentx/tree/2.3/radio/src/telemetry
crossfire.cpp / crossfire.h

c. Inav https://github.com/iNavFlight/inav/tree/master/src/main/telemetry
crsf.c / crsf.h

d. Betaflight https://github.com/betaflight/betaflight/tree/master/src/main/telemetry
crsf.c / crsf.h

Saxin, 2019-06-19:
Captured data using Realterm,

[image]

[capture1.txt](https://github.com/raul-ortega/u360gts/files/3304476/capture1.txt)

Saxin, 2019-07-15:
Adding Inav BB log files and captured CRSF mirror log, captured via. bluetooth (app on phone)
The btlog file contains both flights in sequence.

[btlog (2).log](https://github.com/raul-ortega/u360gts/files/3392702/btlog.2.log)
[LOG00001.TXT](https://github.com/raul-ortega/u360gts/files/3392703/LOG00001.TXT)
[LOG00002.TXT](https://github.com/raul-ortega/u360gts/files/3392704/LOG00002.TXT)

raul-ortega, 2019-08-06:
I'm working on this. Please, could you give to me some reference about lat/lon/altitude where data were captured?

raul-ortega, 2019-08-06:
Frame types and values parsed:

GPS 31914328
GPS 34971280
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914328
GPS 34971280
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914328
GPS 34971279
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914328
GPS 34971279
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914328
GPS 34971279
GSpd 0
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971278
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971278
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971278
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971277
GSpd 0
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971277
GSpd 0
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971277
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971276
GSpd 0
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971276
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971276
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971276
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971276
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914327
GPS 34971276
GSpd 0
Hdg 0
Alt 0
Sats 19
GPS 31914326
GPS 34971276
GSpd 0
Hdg 0
Alt 0
Sats 19
GPS 31914326
GPS 34971276
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914326
GPS 34971276
GSpd 1
Hdg 0
Alt 0
Sats 18
GPS 31914326
GPS 34971277
GSpd 2
Hdg 0
Alt 0
Sats 19
GPS 31914326
GPS 34971277
GSpd 0
Hdg 0
Alt 0
Sats 19
GPS 31914326
GPS 34971277
GSpd 0
Hdg 0
Alt 0
Sats 19
GPS 31914325
GPS 34971277
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914325
GPS 34971277
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914325
GPS 34971277
GSpd 0
Hdg 0
Alt 0
Sats 19
GPS 31914325
GPS 34971277
GSpd 3
Hdg 0
Alt 0
Sats 19
GPS 31914325
GPS 34971277
GSpd 3
Hdg 0
Alt 0
Sats 19
GPS 31914325
GPS 34971277
GSpd 1
Hdg 0
Alt 0
Sats 20
GPS 31914325
GPS 34971278
GSpd 3
Hdg 0
Alt 0
Sats 20
GPS 31914325
GPS 34971278
GSpd 2
Hdg 0
Alt 0
Sats 20
GPS 31914325
GPS 34971278
GSpd 1
Hdg 0
Alt 0
Sats 20
GPS 31914324
GPS 34971279
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914324
GPS 34971279
GSpd 1
Hdg 0
Alt 0
Sats 19
GPS 31914324
GPS 34971279
GSpd 1
Hdg 0
Alt 0
Sats 20
GPS 31914324
GPS 34971279
GSpd 1
Hdg 0
Alt 0
Sats 20
GPS 31914323
GPS 34971279
GSpd 0
Hdg 0
Alt 0
Sats 20
GPS 31914323
GPS 34971279
GSpd 3
Hdg 0
Alt 0
Sats 19
GPS 31914322
GPS 34971279
GSpd 1
Hdg 0
Alt 0
Sats 20
GPS 31914322
GPS 34971279
GSpd 1
Hdg 0
Alt 0
Sats 20
GPS 31914321
GPS 34971279
GSpd 2
Hdg 0
Alt 0
Sats 20
GPS 31914321
GPS 34971279
GSpd 1
Hdg 0
Alt 0
Sats 20

shannonbaker, 2019-08-06:
Saxon is from Israel.
Jerusalem GPS coordinates are:
Latitude: 31° 46' 8.54" N
Longitude: 35° 12' 58.79" E

This may assist you Raúl.

raul-ortega, 2019-08-06:
[image]

Saxin, 2019-08-06:
@Raul the location is pin point accurate

raul-ortega, 2019-08-06:
Please provide a new capture, I need altitude values greater than 0

raul-ortega, 2019-08-06:
@Saxin please attach a jpg/png file with a schema about hardware used. Are you using tarnis sport connecto, with orel without inverted?

shannonbaker, 2019-08-06:
@Raul, altitude from a flight controller sets the zero from takeoff point usually at arming.  I suspect this is what is captured in the data you demonstrated above.  Saxin would need to provide full logs of a flight from takeoff to see altitude change.

Saxin, 2019-08-06:
@raul-ortega Hi and many thanks for pushing this, I'm sure many will much appreciate your work on this.

1. Scheme - will do.
2. Taranis - battery bay UART port (right connector), UART TX -> Inverter -> BL Adapter RX.
3. Set BL baud to 115200.
4. Opentx 2.2.4 (must), Enable smart port mirror in global menu (despite the confusing name CRSF is being mirrored).

The flight was long, but somehow telemetry link was lost / disconnect hence the short log captured.
Due to recent addition to our family I'm won't be able to provide new capture in the upcoming few weeks.

raul-ortega, 2019-08-06:
@Saxin Congratulstions!
Thanks for the information, I'll tell a friend to capture more data.

sniperxnl, 2019-08-06:
Guys my Horus x10s have also build in Bluetooth, possible to use this one to connect to the tracker?

Or is it possible to transmit data from crsf >ble>433 and build a 433 on the tracker to get more distance between tracker and crossfire

raul-ortega, 2019-08-09:
[image]

Saxin, 2019-08-10:
@raul-ortega - schematic added to https://github.com/raul-ortega/u360gts/issues/47#issue-457677113

putimir, 2019-08-19:
> Guys my Horus x10s have also build in Bluetooth, possible to use this one to connect to the tracker?

I'm also interested in this solution - is it possible? Thanks for your time and efforts!

raul-ortega, 2019-08-19:
@putimir in order to be able to display crossfire telemetry in your Horus x10s, it is needed the [TBS inverter mod](https://blog.seidel-philipp.de/fixed-inverter-mod-for-tbs-crossfire-and-frsky-qx7/).

Onces this telemetry is well received and displaye, you have to figure out the way to relay it out from your Taranis. For Taranis X9D, when is upgraded to opentx 2.2.4, it allows to mirror telemetry throught S.port connector. I don't now if Horus X10S uses the bluetooth as S.Port mirroring, or you have to add extra mod bluetooth for your taranis.

putimir, 2019-08-19:
Yes of course, I have telemetry well working, fortunately I'm one of the few lucky ones with very early units, that don't need the mod :D. I get all telemetry perfect @ 115000 baud...
The X10 has a module installed:

[image]

, but I don't know how to work it, there is nothing in the openTX tha relates to Bluetooth

raul-ortega, 2019-08-19:
@putimir So, if you can see crossfire telemetry on your display and your radio is updated with opentx 2.2.4, try changing bauds to 115200 and try to capture a log file from android with Blueterm, for example, and attach the fail to a anew post.

putimir, 2019-08-19:
@raul-ortega: thanks, will try.

@anybody using X10: are there any settings regarding bluetooth in OpenTX? I suppose help from Frsky is out of the question haha

raul-ortega, 2019-08-19:
@putimir unfortunately no developer in this project owns a X10, what we know about it is what I have mentioned before, the "inverter mod" might be necessary depending on your radio, and opentx 2.2.4 allows mirroring on aux s.port, and 115200 bauds are needed. We do not know if the bluetooth of your x10 is capable of mirroring crossfire telemetry or you need some extra hardware mod. I think that the hardware topic is better to be treatted in specific forum for Horux X10 and/or Crossfire, others users might help you to solve it.

I have been investigating in others github threads and tested several log files, some of then from Horux x10 users, and the parser I'm developing did not found GPS frames whithin the data.

putimir, 2019-08-19:
Sorry, it seems we were typing at the same time, take a look at my edit post above...I will try to capture the stream and report.

raul-ortega, 2019-08-19:
@putimir those are good news! Please attach anything you can capture, I'll parse it in a few minutes...

raul-ortega, 2019-08-19:
@putimir remember to change bauds to 115200

putimir, 2019-08-19:
Yes, did notice that, thanks

raul-ortega, 2019-08-19:
Do not worry about it, the information you are posting might be interesting for other users.

putimir, 2019-08-19:
Hello, I now updated the radio to 2.2.4, enabled BT in Telemetry mode, but I just cannot pair my smartphone (Pocophone F1, Pie)...it always gives me "incorrect pin" (which is 0000, and cannot be changed)...Ideas?

putimir, 2019-08-19:
@Saxin 

> 4\. Enable smart port mirror in global menu

Hi, do you know, is this setting also present in Horus radios? I swear I can't find it :|

Saxin, 2019-08-19:
@putimir  - this is located in radio menu (global) of taranis X9D+, I don't have a Horus so can't confirm for you.

putimir, 2019-09-05:
Hello, just to follow up:

To recap: I was trying to capture data from my Android device, via BT link, on Horus X10S with a Crossfire Micro TX module. After turning on the built-in BT module on the Horus, and setting it to telemetry, although the TC was seen while scanning, none of the apps/debug tools captured anything; I first thought it was me having BT connectivity issues. I found some conversations staating, that only s.port/fport telemetry is being pushed to the Horus BT adapter. 

So I setup a s.port test rig and behold, the telemetry is pouring in via BT (i'm using the bt_debug-debug.apk.zip from https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/issues/4#issuecomment-480041038), but not if I'm connected to a XF rig.

So, I think I can confirm, that the OpenTX (probably?) is not pushing the telemetry to the BT...

If I want to do this, what should I do? Do I file a feature  request/issue on the OpenTX git? Is there a person I can talk to directly?

Thanks a lot for all your former and future efforts. Thank you!

Saxin, 2019-09-05:
@putimir This limitation is brobably sw related on Opentx side, I suggest opening an issue on opentx github and hopefully they will agree to update

putimir, 2019-09-05:
Ok, thanks, filed an issue here: https://github.com/opentx/opentx/issues/6712#issue-489609551, in case anyone else is interested in this.

putimir, 2019-09-05:
Wow, SPEEDY GONZALES :D 

https://github.com/opentx/opentx/pull/6713

raul-ortega, 2019-09-05:
Yeah, quite fast!

putimir, 2019-09-06:
YESS! Getting data via internal BT on the Horus X10, with TBS Crossfire Micro module! 

I'm using 115200 baud on the XF module. Will try later with 400000 baud. 

Here is the captured data, @raul-ortega, if you will take a look if it is sane....there should be a fixed position in there, 46...., 14....

Thanks!
[btlog.log](https://github.com/raul-ortega/u360gts/files/3582398/btlog.log)

raul-ortega, 2019-09-06:
@putimir  That log has more than 2800 records (see attached).  The log starts with 7 sats and good gps position. Then sats and position go to 0, it seems like a restart. And finally sats and fix get new values (sats value goes from 4 to 9):

[putimir_btlog.txt](https://github.com/raul-ortega/u360gts/files/3585878/putimir_btlog.txt)

[image]

raul-ortega, 2019-09-06:
> 
> 
> YESS! Getting data via internal BT on the Horus X10, with TBS Crossfire Micro module, OpentTX, nightly build 2.3.0N80!
> 
> I'm using 115200 baud on the XF module. Will try later with 400000 baud.
> 
> Here is the captured data, @raul-ortega, if you will take a look if it is sane....there should be a fixed position in there, 46...., 14....
> 
> Thanks!
> [btlog.log](https://github.com/raul-ortega/u360gts/files/3582398/btlog.log)

400000 is not supported. Max baud supported is 250000.

putimir, 2019-09-06:
> @putimir That log has more than 2800 records (see attached). 

Wow! It was running only a minute or two, I think ...

putimir, 2019-09-19:
Happy to report, that I have a functioning system, telemetry pouring in @10-15Hz rate: 
[image]

raul-ortega, 2019-09-19:
Bravo!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

raul-ortega, 2019-10-15:
Related mods for Taranis QX7 users:

[Taranis QX7 Bluetooth mod with OpenTX](https://github.com/raul-ortega/u360gts/blob/master/docs/Taranis_QX7_Bluetooth_mod_with_OpenTX.pdf)

[How to Convert QX7 to QX7S](https://github.com/raul-ortega/u360gts/blob/master/docs/How_to_Convert_QX7_to_a_QX7S.pdf)

redjacketKR, 2019-10-15:
Thank you for the mod for QX7 and Taranis users. I am using Jumper T16 and I saw a guy attached X10S BT module to Jumper T16. Let's say X10S BT and X12S BT is same and I can attach it to Jumper. Then will it work like QX7?

So I have to prepare following hardware
1) X12S BT for Jumper T16
2) HM-10 module for U360GTS  

If we're gonna use the same openTX 2.3.1~ then it supposed to be working right?
(I think I will asked the same question to JUMPER T16 community)

skaman82, 2020-02-14:
@Saxin I have made a PCB based on you shematic for everyone to order: https://oshpark.com/shared_projects/V3HPfIeN

edugit, 2020-02-22:
@Saxin @skaman82 
Hi Saxin,
I am trying to recopilate all the components to make the inverter with the schematic you kindly posted.
I can't find the regulator with the reference that appears in the schematic.
 Could it be a typo error and it is "ams 1117" instead of "ams 1177"?
Thanks

skaman82, 2020-02-23:
Yes it is ams1117:https://m.reichelt.de/ldo-regler-fest-5-v-sot-223-lm-1117-mp5-0-p109330.html?PROVID=2788&gclid=Cj0KCQiA4sjyBRC5ARIsAEHsELEySpSHrokl2LqXw4gx0twhpQh3vJKs-tlqNSDpQBe0i-dqD__-kVQaAi74EALw_wcB&&r=1Von meinem Huawei-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: edugit <notifications@github.com>Datum: Sa., 22. Feb. 2020, 18:04An: raul-ortega/u360gts <u360gts@noreply.github.com>Cc: Albert Kravcov <albert@wd-design.de>, Mention <mention@noreply.github.com>Betreff: Re: [raul-ortega/u360gts] Crossfire native telemetry support, CRSF relay to u360gts tracker (#47)@Saxin @skaman82
Hi Saxin,
I am trying to recopilate all the components to make the inverter with the schematic you kindly posted.
I can't find the regulator with the reference that appears in the schematic.
Could it be a typo error and it is "ams 1117" instead of "ams 1177"?
Thanks

edugit, 2020-02-23:
Thank you very much!!!!!!

Fpv4tw, 2020-05-22:
Though not reliable to some users, can you still make the crossfire module bluetooth connection available as well? Thank you.

skaman82, 2020-05-26:
@Fpv4tw it is available already (MAVLINK)

Fpv4tw, 2020-05-26:
@skaman82 Cool! So I just need to select MAVLINK Telemetry and pair a bluetooth module (connected to UART1) to the crossfire bluetooth, that's all? AWESOME!

skaman82, 2020-05-26:
@yes, also set your CRSF build-in module to Mavlink Emu

Fpv4tw, 2020-06-05:
@skaman82 Thank you very much!

raul-ortega, 2020-09-25:
I close this issue because CRSF protocol is already implemented and working since release [11.0.0](https://github.com/raul-ortega/u360gts/releases/tag/11.0.0).

closed 2020-09-25 by raul-ortega