raul-ortega/u360gts #49: FrSky X Passthrough (APM10): Altitude issue

raul-ortega, 2019-07-22:
Hi Raul,

I tried and set telemetry_altitude_priority = BARO, but it does not work.
Give us a way to fix tilt servo to follow the model. Тhe speed of BT is
9600 from Taranis x9d.

На пн, 22.07.2019 г. в 0:38 ч. Raúl Ortega <notifications@github.com>
написа:

> Please, in CLI try this:
>
> set telemetry_provider=INAV
> save
>
> —
> You are receiving this because you are subscribed to this thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/raul-ortega/u360gts/issues/48?email_source=notifications&email_token=AF6JA52HQDFOZHTTRNHWQODQATJNXA5CNFSM4IC7ZQJKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2OMD2Q#issuecomment-513589738>,
> or mute the thread
> <https://github.com/notifications/unsubscribe-auth/AF6JA54I2MTS7DVBPYMVKELQATJNXANCNFSM4IC7ZQJA>
> .
>

# version
# u360gts / NAZE 10.1.0 May 18 2019 / 09:31:41 (eeeac27)

# serial
serial 0 1 115200 57600 0 115200
serial 1 2 115200 9600 0 115200

# feature
feature VBAT
feature SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature DISPLAY
feature EASING
feature -NOPID
feature -DEBUG
feature EPS
feature AUTODETECT

# set
set looptime=1000
set emf_avoidance=OFF
set rssi_channel=0
set rssi_scale=30
set rssi_zoom=35
set servo_pwm_rate=50
set gps_baud=4
set gps_provider=UBLOX
set gps_sbas_mode=AUTO
set gps_auto_config=ON
set gps_auto_baud=OFF
set gps_min_sats=6
set gps_home_beeper=ON
set telemetry_inversion=OFF
set battery_capacity=0
set vbat_scale=110
set vbat_max_cell_voltage=46
set vbat_min_cell_voltage=33
set vbat_warning_cell_voltage=35
set align_gyro=DEFAULT
set align_mag=DEFAULT
set align_board_roll=0
set align_board_pitch=0
set align_board_yaw=0
set gyro_lpf=42
set magzero_x=50
set magzero_y=-898
set magzero_z=-385
set p=2500
set i=20
set d=250
set max_pid_error=10
set max_pid_accumulator=5000
set max_pid_gain=500
set pid_divider=15
set pan_pin=0
set pan0=1498
set pan0_calibrated=1
set pan_calibration_pulse=1400
set min_pan_speed=60
set offset=0
set offset_trim=10
set init_servos=1
set tilt_pin=1
set tilt0=1050
set tilt90=2000
set tilt_max_angle=0
set easing=2
set easing_steps=60
set easing_min_angle=4
set easing_millis=15
set telemetry_baud=8
set telemetry_protocol=128
set telemetry_min_sats=6
set telemetry_provider=APM10
set telemetry_home=AUTO
set start_tracking_distance=4
set start_tracking_altitude=2
set mag_declination=0
set mag_calibrated=1
set min_logic_level=60
set nopid_min_delta= 0.200
set nopid_map_angle=90
set nopid_max_speed=200
set eps=3
set eps_distance_gain=100
set eps_heading_gain=50
set eps_speed_gain=60
set eps_frequency=250
set eps_interpolation=OFF
set eps_interpolation_points=3
set update_home_by_local_gps=OFF
set max_speed_filter=0
set telemetry_altitude_priority=GPS

_Originally posted by @tsvetanminchev in https://github.com/raul-ortega/u360gts/issues/48#issuecomment-513814488_

raul-ortega, 2019-07-22:
Please, try this new version: [dev 10.0.1](https://github.com/raul-ortega/u360gts/raw/master/other-releases/amv-open360tracker_dev10.1.1.zip).

Unzip it and flash the hex file for your board.

I tested it with real data, I placed my Nimbus 1800 VTOL on the balcony of my house and the tilt moved when the altitude and distance reached the limits.

tsvetanminchev, 2019-07-24:
Version 10.1.1 works with APM😀 Protocol 10 better than expected. I tried everything for a whole year and it's already successful. Well done!!!!

tsvetanminchev, 2019-07-24:
I use  this telemetry on Taranis X9D (https://github.com/yaapu/FrskyTelemetryScript#features) and pass to tracker via BT . My plane has Omnibus F4 with version 3.9.8 and copter 3.6.9 version. There are small deviations in the tilt position but this will clear it with settings. Thanks.

raul-ortega, 2019-07-24:
Congratulations!
Thanks so much for your feedback!

closed 2019-07-24 by raul-ortega