raul-ortega/u360gts #60: No Mavlink-out when LTM-in

MFlghts, 2019-08-01:


raul-ortega, 2019-08-04:
The LTM capture is ok. I have tested it with Bluepill and everything seems to be ok as shown in this 
[image].

raul-ortega, 2019-08-05:
Sp Racing F3 also tested, in this case connecting the bluetooth to softserial 2. It seems to be working with official version 10.1 (10.1.3).

[image]

raul-ortega, 2019-08-05:
I did the spracingf3 test with a bec powering servo rail. The bluetooth vcc pin is connected to + pin of servo rail. I connected gnd, rx and tx to softserial 2 (serial 31), as shown in the video.

raul-ortega, 2019-08-05:
This is my configuration file.
[u360gts_config_10.1.3_spracingf3_ltm2mavlink_softserial31.txt](https://github.com/raul-ortega/u360gts/files/3465513/u360gts_config_10.1.3_spracingf3_ltm2mavlink_softserial31.txt)

raul-ortega, 2019-08-07:
Solved! User configured tracking distance as 0, and this caused tracking to do not start.

closed 2019-08-07 by raul-ortega

MFlghts, 2019-08-07:
1. User configured **tracking ALT** - not distance as 0.
2. Tracking does start with ALT set to 0, but telemetry relay is not.
3. This is a bug, because 0 - means that both tracking and relay should work at START TRACKING ALT = 0. 

I don't think issue is closed.