raul-ortega/u360gts #66: cc3d: usb connection over standard cc3d usb port

Maarten112, 2019-09-28:
I found an old cc3d, great for this project. 

However, I noticed an issue with the usb connection. 

**Problem:**
it's not possible to connect the cc3d to pc using the standard usb port from the cc3d. Windows gives the note: USB device not recognised.

**What Did I do?**
flashing amv-open360tracker_10.1.3_CC3D.hex using an ftdi, because of the lack of a 2.1 USB-DFU interface as it is an STM32F1 based FC.

**Other things I tried:**
- Flash betaflight 3.2 (latest version that supports cc3d).  normal USB port of cc3d **is working** with Betaflight.
- Flash latest cleanflight (2.5.0). normal USB port of cc3d **is not working** with cleanflight

Since u360gts uses cleanflight as a basis for u360gts, I suspect something is wrong inside cleanflight. I will also try to have a look at is soon to see if I'm able to fix the issue.

lucafedechen, 2019-11-05:
hello 
i have one cc3d new and never used!!
you have a schematic for proper pinout connection in cc3d to run u360gts??

gstyle97, 2020-02-22:
You have to use an ftdi programmer!

http://thrustworx.com/how-to-install-cleanflight-cc3d-guide/

gstyle97, 2020-02-22:
#73