raul-ortega/u360gts #68: Feature Request: Allow Telemetry on UART3 on SPRacingF3 target

skaman82, 2020-02-13:
UART1 is shared with USB and it is a real hassle to configure the tracker. I have to physically remove the telemetry wire before i can use the configurator.

raul-ortega, 2020-02-23:
Take a look here: [https://www.facebook.com/groups/u360gts/search/?query=uart3&epa=SEARCH_BOX](https://www.facebook.com/groups/u360gts/search/?query=uart3&epa=SEARCH_BOX)

skaman82, 2020-02-23:
@raul-ortega sorry, I meant UART2 not3

raul-ortega, 2020-02-23:
Ok, let's study it... I'll try to allow the possibility to use any port for ingoing telemetry. But take into account that telemetry and configurator will always use the same uart...

skaman82, 2020-02-23:
That is actually the point to use a different uart for telemetry and at the same time to be able to use USB for msp for configurator. Would it be possible?

raul-ortega, 2020-02-23:
We don't use MSP.

raul-ortega, 2020-02-23:
There is something that might be getting you into confusion, the fc may
only be configured while in CLI mode, so you may not send telemetry at the
same time from a different uart. To do that a lot of code should be changed.

skaman82, 2020-02-23:
My main problem with the shared uart is: if I want to configure the tracker using configurator, I have to Jake it apart and disconnect the tx wire from my BT module. If I don't do this configuration via configurator won't work. Do you have any better solution in mind?

raul-ortega, 2020-02-23:
A bluetooth module connected to your PC through a usb-ttl adapter, that's
what I do. Other people use micro toggle switch and pin headers.

arf-it, 2020-04-05:
> A bluetooth module connected to your PC through a usb-ttl adapter, that's
> what I do. Other people use micro toggle switch and pin headers.

Now I understand why there is a toggle on the bluepill schematic!

skaman82, 2020-04-06:
> A bluetooth module connected to your PC through a usb-ttl adapter, that's what I do. Other people use micro toggle switch and pin headers. El dom., 23 feb. 2020 a las 22:37, Albert Kravcov (<notifications@github.com>) escribió:
> […](#)

I figured out a different solution that won't require a toggle switch. I just installed a shokkty diode on the telemetry line and the usb is working fine now without the need to disconnect the telemetry line.

closed 2020-05-19 by skaman82