raul-ortega/u360gts #71: Blank screen after changing telemetry provider

skaman82, 2020-02-18:
When telemetry was set to auto, changing telemetry provider on the OLED results in a blank screen instead of the root menu.

[image]

raul-ortega, 2020-02-23:
I tried to reproduce the issue without success! I'm using develompment version 11.0.1 ([download](https://github.com/raul-ortega/u360gts/raw/master/other-releases/amv-open360tracker_dev11.01.zip))

[https://www.youtube.com/watch?v=utINnAOf7Og](https://www.youtube.com/watch?v=utINnAOf7Og)

skaman82, 2020-02-23:
I am using the current release version. Will try the dev version

skaman82, 2020-06-22:
i never experienced this issue since, so I am closing this one

closed 2020-06-22 by skaman82