raul-ortega/u360gts #73: CC3D board pin mapping

gstyle97, 2020-02-22:
Hi,

i'm trying to use an **cc3d** board for u360gts,

but i have a lot of problems because there are no **CC3D pinout map**.

I've found:

pwm3 = tilt
pwm4 = pan

but i need :
**TX-RX for bluethoot
i2c for magometro and display
2 pins for the buttons
TX-RX for gps**

--------------------------------------------------------------------------------------------------------------

This are the info of the board that i've found 

Pinouts
The 8 pin RC_Input connector has the following pinouts when used in RX_PPM/RX_SERIAL mode

Pin	Function	Notes
1	Ground	
2	+5V	
3	PPM Input	Enable feature RX_PPM
4	SoftSerial1 TX / Sonar trigger	
5	SoftSerial1 RX / Sonar Echo	
6	Current	Enable feature CURRENT_METER. Connect to the output of a current sensor, 0v-3.3v input
7	Battery Voltage sensor	Enable feature VBAT. Connect to main battery using a voltage divider, 0v-3.3v input
8	RSSI	Enable feature RSSI_ADC. Connect to the output of a PWM-RSSI conditioner, 0v-3.3v input
The 6 pin RC_Output connector has the following pinouts when used in RX_PPM/RX_SERIAL mode

Pin	Function	Notes
1	MOTOR 1	
2	MOTOR 2	
3	MOTOR 3	
4	MOTOR 4	
5	LED Strip	
6	Unused	
The 8 pin RC_Input connector has the following pinouts when used in RX_PARALLEL_PWM mode

Pin	Function	Notes
1	Ground	
2	+5V	
3	Unused	
4	CH1	
5	CH2	
6	CH3	
7	CH4/Battery Voltage sensor	CH4 if battery voltage sensor is disabled
8	CH5/CH4	CH4 if battery voltage monitor is enabled
The 6 pin RC_Output connector has the following pinouts when used in RX_PARALLEL_PWM mode

Pin	Function	Notes
1	MOTOR 1	
2	MOTOR 2	
3	MOTOR 3	
4	MOTOR 4	
5	Unused	
6	Unused	
Serial Ports
Value	Identifier	Board Markings	Notes
1	VCP	USB PORT	
2	USART1	MAIN PORT	Connected to an MCU controllable inverter
3	USART3	FLEX PORT	
4	SoftSerial	RC connector	Pins 4 and 5 (Tx and Rx respectively)
The SoftSerial port is not available when RX_PARALLEL_PWM is used. The transmission data rate is limited to 19200 baud.

To connect the GUI to the flight controller you just need a USB cable to use the Virtual Com Port (VCP) or you can use UART1 (Main Port).

CLI access is only available via the VCP by default.

Main Port
The main port has MSP support enabled on it by default.

The main port is connected to an inverter which is automatically enabled as required. For example, if the main port is used for SBus Serial RX then an external inverter is not required.

Flex Port
The flex port will be enabled in I2C mode unless USART3 is used. You can connect external I2C sensors and displays to this port.

You cannot use USART3 and I2C at the same time.

Flex port pinout
Pin	Signal	Notes
1	GND	
2	VCC unregulated	
3	I2C SCL / UART3 TX	3.3v level
4	I2C SDA / UART3 RX	3.3v level (5v tolerant

atomiclama, 2020-02-23:
Been pointed here from face book.
Got this some years ago from ardupilot website.
Hope this is of some use
[CC3D Schematic.pdf](https://github.com/raul-ortega/u360gts/files/4240762/CC3D.Schematic.pdf)
.

arf-it, 2020-04-04:
> Hi,
> 
> i'm trying to use an **cc3d** board for u360gts,
> 
> but i have a lot of problems because there are no **CC3D pinout map**.
> 
> I've found:
> 
> pwm3 = tilt
> pwm4 = pan
> 
> but i need :
> **TX-RX for bluethoot i2c for magometro and display 2 pins for the buttons TX-RX for gps**

i2c for magometro and display: pin 3 and 4 of the flex port (tested and working)
still searching for GPS,BLUETHOOT and buttons..

WizzX-FPV, 2021-07-02:
I'm also trying to reuse old CC3D for tracker.

**Just try to summarize all info here:**

- It is not possible to use CC3D USB for programming and configuration,

instead I'm using ST Flash loader and Arduino Nano as programmer.
guide here:
[](http://sgeorgiev.com/blog/tag/openpilot-cc3d/)
[image]

- PINOUT
I can confirm all  connection below works (but not tested Bluetooth till now)

Main Port
```
1 GND
2 VCC unregulated
3 UART1 TX **Bluetooth**
4 UART1 RX **Bluetooth**
```

Flex Port
```
1 GND
2 VCC unregulated
3 I2C SCL - **Magnetometer + OLED**
4 I2C SDA - **Magnetometer + OLED**
```

RXport
```
1 Ground
2 VCC unregulated
3 
4 SoftSerial1 TX **GPS** (softserial 30)
5 SoftSerial1 RX **GPS** (softserial 30)
6 
7 Battery Voltage 0v-3.3v input
8 RSSI 0v-3.3v input
```

Pin Function Notes
```
1 **home button** (GND+sig short)
2 **calibration button**
3 **Servo - tilt**
4 **Servo - pan**
5 
6 
```

- CLI
have to set softserial for GPS
```
serial 30 2 115200 9600 115200 115200
save
```

and than set GPS:
```
feature GPS
set gps_provider=UBLOX
set gps_auto_config=ON
save
```
sometimes it not accept for a first time, so you need to try more times

shro70, 2021-07-22:
Thank you I was struggling to connect GPS and Rssi