raul-ortega/u360gts #74: QMC5883 HMC5883L mag not working

L4ky, 2020-02-26:
Hi, i've read in the docs that DA package is not supported yet.
Unfortunately i discovered about this limit only today that i received my 2 mags from ebay.

Those are sold as "GY-271 QMC5883 HMC5883L" and the package has "DB 5883 8001".
I think they're the same as DA package, but i don't know the differences.

Is there any chance that it gets supported by u360gts?
Maybe porting the QMC5883 driver from INAV is a easier way?

If i can help somehow with improving u360gts, just ask!

Thank you
Luca

raul-ortega, 2020-02-26:
Any contribution in that regard will be welcome.
Contributing is guite easy, yo should clone the repo, add code, build
binaries, test ... and if everything works send a pull request.

L4ky, 2020-02-26:
I don't think i'm skilled enough to make it work. 

Do you have any advice for me?
Are you going to work on it soon?

L4ky, 2020-02-29:
I worked on it.
Got it working.

See https://github.com/raul-ortega/u360gts/pull/75

raul-ortega, 2020-03-02:
I'll take a look to the PR today...

arf-it, 2020-04-20:
I'm trying the development build with a SP Racing F3 and so far it's working correctly:
the official release was not able to detect the onboard gyro while the development build is working properly.
So far I tested the gyro calibration and some ground test (cannot fly 'cause quarantine).

raul-ortega, 2020-09-25:
Release [11.1.0](https://github.com/raul-ortega/u360gts/releases/tag/11.1.0) solves this issue.

closed 2020-09-25 by raul-ortega