raul-ortega/u360gts #80: Taranis X9D+ 2019 and R9M2019 ACCESS

zosko, 2020-03-08:
Is this tracker working with Taranis X9D+ 2019 and R9M2019 external module ?
I have problem to make tracker, i have try ghetto tracker but that not support S.Port. Currently i have issues to get S.Port telemetry from my taranis, because is different from old taranis.

There are working examples with old R9M and old taranis but on new version that is problem i think.

Is any have tested on new taranis and r9m access is it working ?

L4ky, 2020-03-09:
How are you transmitting the telemetry to your tracker?

I'm using a R9M with ACCST Flex Firmware. It works fine with U360GTS.

zosko, 2020-03-09:
I use ACCESS LBT, maybe problem is ACCESS not allow SPort anymore, maybe i can downgrade R9M2019 to ACCST firmware and try it.

skaman82, 2020-05-19:
Same problem here. I have an R9M Lite Pro and it comes only in ACCESS version. Telemetry not working with that module. As soon I pop in the crossfire, it is working fine. So I guess it is not working with R9M (ACCESS) because FrSky-Kompatibilitätstests changes some things in the telemetry stream. I would really appreciate if support for ACCESS telemetry would be added.

I collected some data if that helps:
Telemetry Logic capture at 57600 baud and serial monitor output:
[image]
[Logic.zip](https://github.com/raul-ortega/u360gts/files/4651988/Logic.zip)

Also I've found out that this App is able to read my ACCESS telemetry stream: https://github.com/CrazyDude1994/android-taranis-smartport-telemetry?fbclid=IwAR0HFt5bSc1tBTD5SEV3RenQuxW_0wtxeLIOn3rF4jxECHOHBcpcQnxj8Bo
[image]

zosko, 2020-05-19:
Please log data on 115200 and 460800 and put here.

skaman82, 2020-05-19:
@zosko the HM10 can't do 460800 - so a will put a 115200 log in here

skaman82, 2020-05-19:
@zosko 115200 log and my sensors list (flex has one sensor more)
[115200_log.zip](https://github.com/raul-ortega/u360gts/files/4652804/115200_log.zip)

skaman82, 2020-05-29:
@zosko any progress on this? That also might help: https://www.rcgroups.com/forums/showpost.php?p=44636915&postcount=1005

zosko, 2020-05-29:
@skaman82  no success, but i have made mu own tracker with ground station
https://github.com/zosko/iNavTelemetry
https://github.com/zosko/R9M_Inav_antenna_tracker

but however,  see that code, from RC group i will make a test, seems that will be solution :)

skaman82, 2020-06-12:
Some new findings: 
ACCESS Documentation (PXX2 Protocol): [ACCESS.protocol.docx](https://github.com/raul-ortega/u360gts/files/4771191/ACCESS.protocol.docx)

• S.PORT pin is not used anymore for communication. Instead PXX pins are used. 
Communication speed is **450000 baud (inverted)**. 

• INTERNAL Bluetooth seems to transfer data differently on my x-lite **at 115200 baud** - still working fine with https://github.com/CrazyDude1994/android-taranis-smartport-telemetry seems the old protocol - but it is **not working with the u360gts unfortunately** 

I have logged the communication protocols of both (PXX and internal Bluetooth UART).
Here are the results: [BT-and-PXX.logicdata.zip](https://github.com/raul-ortega/u360gts/files/4771346/BT-and-PXX.logicdata.zip)
[image]

skaman82, 2020-06-12:
About the Internal BT Module - I found a quick DIRTY fix. It seems that CRC has changed. Due to the failed CRC checks in frskyx.c all data is blocked. After commenting this out:

`  if (!checkSportPacket(packet)){
	  telemetry_failed_cs++;
	  //return; < ignore failed crc an go on
  }`

it seems to work fine. So it needs just an alternative CRC check method. I will raise a separate issue for this.

MrCoopa, 2020-10-11:
Is it actually working? I want to build a tracker with R9M2019 Access but I don't know if it get fixed.

closed 2021-04-15 by zosko