raul-ortega/u360gts #91: Alternative CRC check for R9 Telemetry via build-in BLE module is needed

skaman82, 2020-06-12:
I can't get my Build-in BT module to work with the Tracker FrskyX Telemetry CRC 
is checked here currently in frskyx.c:

```
 if (!checkSportPacket(packet)){
	  telemetry_failed_cs++;
	  return;
  }
}
```
after removing the "return;" it seems to work just fine. Pleas add an alternative CRC routine that is working with the newer R9 and Build-In BT modules. Baudrate from the build-in module is 115200 btw. Here some input:

[ACCESS.protocol.docx](https://github.com/raul-ortega/u360gts/files/4773583/ACCESS.protocol.docx)
[Buld-in-Bluetooth.txt](https://github.com/raul-ortega/u360gts/files/4773584/Buld-in-Bluetooth.txt)

janmj, 2020-06-27:
I upgraded my radios from Opentx 2.2.4 to 2.3.9, and then i got this very same problem. So obviously something changed somewhere between these versions. 
I hope this issue can be merged in to master soon, because i really hate to have to downgrade my radios :).
Alternatively i have to build my own version. Is there any good instructions for how to build the U360GTS firmware?

skaman82, 2020-06-27:
@janmj I could send you a hex file for testing. Naze or SprachingF3?

janmj, 2020-06-27:
@skaman82 Thanks, i would appreciate that! I use a NAZE board.

skaman82, 2020-06-27:
@janmj please let me know if that is working
[amv-open360tracker_NAZE.hex.zip](https://github.com/raul-ortega/u360gts/files/4840458/amv-open360tracker_NAZE.hex.zip)

janmj, 2020-06-27:
@skaman82  Thanks a lot, will report back :)

arf-it, 2020-06-28:
> Alternatively I have to build my own version. Is there any good instructions for how to build the U360GTS firmware?
you can use the documents to build inav o cleanflight. I build on my own and if I remember well I just followed those instructions (as a base)
https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Ubuntu.md
it works just fine in windows as well using the WSL.

@skaman82 
It's not clear to me if the CRC issue that you found is related to ACCESS protocol. 
Let me explain: my setup works just fine but it won't if I'm trying to use the internal Bluetooth module with telemetry mirror enabled but I'm still using the "old" ACCST protocol.

janmj, 2020-06-28:
Not using access here, this problem started when i upgraded opentx..
And also was working fine on both internal 2.4 and R9M (not 2019) before the upgrade.

@arf-it Thanks for the tips :)

janmj, 2020-06-28:
@skaman82 I just testet your firmware, and unfortunatly it was even worse :)
With the released firmware at least i got a "connection" with telemetry, meaning the FCS number strted to count. With this i got nothing...

skaman82, 2020-06-28:
@janmj that is strange. I am using the R9 lite Pro module (flex, access) in my x-lite and connecting via build-in BT module. 

@arf-it  BT is using still the old style sport protocol. This setup is working fine with my modified code:
[image]

janmj, 2020-06-28:
Very strange! What version of OpenTx are you on?

janmj, 2020-06-28:
Maybe it's a Horus problem.. I noticed from 2.3.0 the Horus got support for 60 sensors, up from 32 earlier. And i did have to delete and rediscover telemetry on all  my models. Maybe the sensor index is different on the horus...

skaman82, 2020-06-29:
@janmj I am on OpenTX 2.3.9

janmj, 2020-06-29:
@skaman82  Ok, but have you testet on a Horus? I will try to downgrade just to confirm...
But not until the weather get's bad around here :)

skaman82, 2020-06-29:
@janmj OpenTX 2.3.9 is the latest release AFAIK. I don't own a Horus - tested wit x-lite

janmj, 2020-06-29:
Yes, and as i said, my problems began when i upgraded from 2.2.4 to 2.3.9.
But only the Horus had changes to the telemetry in the 2.3.X version(from 32 to 60 possilble sensors).
Thats why i suspect this is the problem, meaning the sensor index is different on the Horus..

brainbubblersbest, 2020-07-11:
Heres the same shit. R9m, x12s. Internal bt mirror.
Telemetry only over two hc05 with inverter.
Happens with 2.3.7 and 2.3.9
ACCST not access

skaman82, 2020-07-12:
@brainbubblersbest you don't need an inverter if using the build-in BT from the radio

skaman82, 2020-07-12:
Since https://github.com/raul-ortega/u360gts/pull/100 is merged, I am closing this for now

closed 2020-07-12 by skaman82

raul-ortega, 2021-04-20:
I have included both methods of CRC calculation, in this way it will no matter is telemetry is extracted directly from TX module or from internal Horus type Bluetooth module.

For test purpouses, please try: [https://github.com/raul-ortega/u360gts/blob/master/other-releases/amv-open360tracker_FRSKY_X_CRC_FIX_ALL_TARGETS.zip](https://github.com/raul-ortega/u360gts/blob/master/other-releases/amv-open360tracker_FRSKY_X_CRC_FIX_ALL_TARGETS.zip)