raul-ortega/u360gts #94: Spontaneous reboots

skaman82, 2020-06-22:
I am experiencing some spontaneous reboots mid flight. Telemetry is coming in via BT (115200) from crossfire @ 10Hz. Either the racker just reboots and sets the new homepoint (somwhere where the Aircraft currently is in the sky) or it enters the CLI mode. Both is pretty unfortunate. 
This happens not often but almost every 4-5th flight. I was able to reproduce it just with the OLED, FC and BT module - so the problem seems not to be electrical. Ver. 11.0.1.
Video: https://youtu.be/Mv7_NI18nV8

skaman82, 2020-06-24:
I noticed the BT module disconnecting and reconnecting from time to time (probably due data overflow or some other reason) and at the time the BT is reconnecting the reboot or cli screen seems to happen.

raul-ortega, 2020-09-25:
Dear @skaman82, has you experienced new reboots after the Fix?

raul-ortega, 2020-09-25:
In your video2 there is not a reboot, it simply enters in CLI mode because string 'RRR' is received. That is what evaluateOtherData function does, it waits string 'RRR', and if found, it enters CLI mode. Your fixes avoid checking 'RRR' string, that's a good solution for this scenario, but not for the reboot one.

skaman82, 2020-09-28:
@raul-ortega no, I had no reboots after my fix

closed 2021-05-09 by raul-ortega