raul-ortega/u360gts #103: Program compiling software

wsqzdz, 2020-12-10:
I want to modify and compile u360gts program again. I want to know what compiler this program is written with. I hope you can give the specific software name

raul-ortega, 2020-12-10:
Please may take a look at:
https://github.com/raul-ortega/u360gts/tree/master/docs/development

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

wsqzdz, 2020-12-11:
Sorry for not reading the document carefully, I plan to try to replace the servo with a stepper motor, thank you very much for your answer

closed 2021-05-01 by raul-ortega