raul-ortega/u360gts #107: Parameter Pan_inverted not respected

AlexC176, 2021-02-01:
SPRacing F3 rc1.1.1.0 sept 23rd 2020 doesn’t respect the parameter pan_inverted = ON|OFF. Makes no difference to incorrect rotation for servo during calibration.

AlexC176, 2021-02-01:
u360gts/src/main/io/serial_cli.c 
Parameter constrained to 0-1500 (midpoint)
Should be 0-3000 to support inverted servos

raul-ortega, 2021-05-02:
Range extended in [this commit](https://github.com/raul-ortega/u360gts/commit/9fbf26f12bafc66f561b6aae508b36bc552ea168). Please, give a try with the appropriate hex file for your board within the [zip file I have uploaded](https://github.com/raul-ortega/u360gts/blob/master/other-releases/amv-open360tracker_dev11.2.0_calibration_pulse_range.zip) just now.

closed 2021-05-03 by raul-ortega