raul-ortega/u360gts #109: CLI command for align mag doesn't stick

AlexC176, 2021-02-10:
Most recent firmware version, align mag command doesn't take effect, I had to manually move the mag within the case to the correct alignment.

Looking at serial_cli.c the line align mag entry is commented out. Was previously raised as an issue, comment was merged into development branch. Looks like change wasn't implemented.

https://github.com/raul-ortega/u360gts/blob/b4e3b760e2748d367b8ddc0f25416860ae6f986a/src/main/io/serial_cli.c#L566

AlexC176, 2021-02-10:
Previous closed issue was https://github.com/raul-ortega/u360gts/issues/17#issue-261029829

raul-ortega, 2021-05-09:
align_mag is unommented in master and development versions since quite time ago.
Please, could you be more explicit about the issue you experienced.