raul-ortega/u360gts #114: Full config loss after power cycle / USB Disconnect

b14ckyy, 2021-06-16:
I just started to look at the U360GTS project again as I want to start to use it regularly now. Installed 11.2.0 RC2 with full chip erase and set up from scratch. 

I had multiple full config losses when I switched from USB to Battery power. It was hard to replicate manually. One time it happened after the following steps: 

- changed settings
- clicked save > system reboots 
- clicked disconnect > system reboots again 
- removed USB > powers off
- Powered on by battery > no display shown
- connected USB > all config lost and all on default

I have then left the new Last Home Safe option off. Since then I had a lot of power cycles and no issue. Can' completely validate yet but it might have something to do with the last home b8ackup function. 

Board: SPRF3; separate BECs for Servos and Board

b14ckyy, 2021-06-16:
Update: Again full config loss just by switching battery power off, and after a few seconds on again.,.. so it is not the last home store option. I have no idea....

ModaFpv, 2022-10-30:
same with you......