raul-ortega/u360gts #125: Servo missing in setup with 2 tilt servos

rueckwaertsflieger, 2023-04-04:
In https://github.com/raul-ortega/u360gts/blob/master/docs/hardware-frame.md a mechanical design from Juan Trillo is referenced, involving a symmetrical 2-tilt-servo setup. That takes care for the otherwise inavoidable torsion of the payload carrier by asymmetrical load. In u360gts a second tilt servo is not defined.

Please introduce a second tilt servo.

juvinski, 2023-04-21:
I'm testing a version customized to work with 2 servos, once inverted with naze 32 controller, are you interested ?

rueckwaertsflieger, 2023-04-21:
> I'm testing a version customized to work with 2 servos, once inverted with naze 32 controller, are you interested ?

Yes, interested. Meanwhile I designed a 2nd servo "replacement", see https://www.thingiverse.com/thing:5944568 , part "027_servo_arm_bearing" . The very bearing is a steel rod with thread just at one end, available as model landing gear axle.

juvinski, 2023-04-21:
I have this branch with the changes.
https://github.com/juvinski/u360gts/tree/servoreverse
I tested on a flip32 board. What board are you using ?
I'm using this design: https://www.thingiverse.com/thing:1182191
I'm still trying to use the u360gts with ardupilot, the only thing I discovered so far is to change the mavlink version to 1 - now all defaults in ardupilot is mavlink 2 what is not supported I believe.

rueckwaertsflieger, 2023-04-22:
I am using CRSF, controller NAZE32. Setup:
Betaflight 4.4 ->
ExpressLRS 3.2.0 with edgeTX 2.8.0
From aux port 2 of radio TX16S via pair of HM-10 to u360gts
After 3 times testing, there's multiple issues left. Half of the occasions that is power reset of the drone, fly and observe tracker, the antenna elevation seems to be stuck to horizontal. So I suspect glitches in the u360gts protocol autodetection and thus will hard configure CRSF for the next flight.
Target is to use https://github.com/ExpressLRS/ExpressLRS/pull/2018, offering a telemetry relay by ELRS TX module with ESP in order to get rid of the HM-10 in transmitter.

rueckwaertsflieger, 2023-04-22:
> I have this branch with the changes. https://github.com/juvinski/u360gts/tree/servoreverse I tested on a flip32 board. What board are you using ? I'm using this design: [...]

Using a NAZE32, input is CRSF via ELRS. Please be aware, I am no Softworker and thus are unable to build a developers environment. Sorry, for testing I need a ready-made binary.

raul-ortega, 2023-04-23:
Official version is running well with [Arduplane 4.3.1 + ExpressLRS 3.0.1 (CRSF protocol)](https://www.youtube.com/watch?v=LcjIwugB3tg)

If you reset the drone, you should also reset the tracker becouse it will try to calculate relative altitude based on drone altitude received before setting up the home position.

Latest RC release implements [Restore Last Known Home Position](https://github.com/raul-ortega/u360gts/releases/tag/11.2.0-RC2) functionality, please read the releases notes, I think it would help to solve some of the issues you make mention to, so you can focus only in your development.

Thanks so much for your contribution.