raul-ortega/u360gts #126: Mag calibration never succeeds with pan reduction gear

rueckwaertsflieger, 2023-04-23:
The design uses a 3:1 pan reduction gear.
When invoking a mag calibration, the calibration run turns the tracker's turret some 180° only. After that, the configurator reports mag calibrated. However, this incomplete run leaves the calibration in an unusable state.
Only a pan calibration does the job, somehow. In case, the calibration start value is short enough, the pan calibration run does make it by turning the turret over more than a full round. However, there typically remains an anglular range, where pan angle error is max and a positioning overshoot happens. The latter working perfectly on most of the other angular range. u360gts reports remaining errors of +/- 2° at max.
I suspect the mag calibration to not perform optimally when done whilst a servo calibration run. Rather, the mag calibration should take care for the geared operation.
Since I have no idea, how the mag calibration run is ruled, I'd like to make two suggestions on probable improvements.
Suggestion 1
In case, calibration is just ruled by a time-constraint, increase time of calibration run. Best was to introduce a pan gear ratio as supposed in https://github.com/raul-ortega/u360gts/issues/40
Suggestion 2
In case, the calibration run is ruled by a sophisticated algorythm, introduce a finish constraint by just detecting two full  360° turns or on full turn at least for the sake of "mission completed". I suggest the criteria to be the second occurrence of all magnetometer axe's maximum whilst turning.

raul-ortega, 2023-04-23:
For a better understanding about mag calibration, please read first [pan servo calibration](https://github.com/raul-ortega/u360gts/blob/master/docs/configuration-pan-servo.md) document. Automatic pan servo calibration proccess first calibrate mag and then continues tryning to figure out the best the stop pulse. But you always can first manually configure the stop pulse using the arrows or writting dwon the value, an then calibrate de mag. Onces the stop pulse si properly configured every time you calibrate de mag it shoudl stop. Finally, from the doc you can deduct mag calibration process is based on spinning time. There are many u360grs based antenna tracker running out there and they work great, no more calibration spinning time seems to be necesary by experience.

rueckwaertsflieger, 2023-04-23:
Thanks for your reply. I will take videos to demonstrate the problem.
Document pan servo calibration is understood.
Status is:
Automatic pan calibration works, including
mag calibration as step one when hitting "calibrate PAN" in configurator. But mag calibration only succeeds, if "Starting Calibration Pulse" is short enough to let the tracker spin fast enough (to gain one full 360° turn at least?). If starting Calibration Pulse is near center stop pulse, mag calibration as part one of pan calibration fails. However, I have a feeling, calibration accuracy could improve.
Hitting "Calibrate Mag" always fails
Will be back with videos.

rueckwaertsflieger, 2023-04-28:
> For a better understanding about mag calibration, please read first [pan servo calibration](https://github.com/raul-ortega/u360gts/blob/master/docs/configuration-pan-servo.md) document. Automatic [...]

Here you find videos showing the behavior when performing a calibration run, and commanding the tracker to turn to main directions north, east, south and west respectively.

01  calibration (around) north: 01calnorth
     https://youtu.be/hiv3l2oBjwo
02 main directions with calibration north: 02maindirec_calnorth
     https://youtu.be/sRw8D1vF7Kw
03calibration south: 03calsouth
     https://youtu.be/y26en6YmzCA
04 main directions with calibration south: 04maindirec_calsouth
     https://youtu.be/PZYC1yfozZM

Findings:
I cannot reproduce the differences between mag calibration run and pan calibration including the mag calibration any more, sorry.
The angle covered during calibration run is less than 180°.
For reference, I put two triangle ruler underneath the tracker. The tips point towards main directions N,E,S,W, aligned with a nautical grade compass.
All main directions as well as calibration run are commanded via configurator. When commanding 90, 180, 270, apparently wild movements of the tracker are caused by typing the angles digit by digit. E.g. typing 270 causes the tracker to start moving towards 2, and 27 degrees before turning towards 270.
Starting condition angle for the calibration runs are around north-90° and south-90° to make sure, north and south are covered during calibration. After calibration, videos 1 and 3, the magnetometer correction angle is tuned to start from main direction north or south, videos 2 and 4.
If north is covered in the angular range during Calibration north, main direction north is reached. All other directions commanded show huge deviations towards north.
Holds true, when covering south in the angular range. Deviations are towards south.
Extreme deviation is near 45° when commanding 90°

rueckwaertsflieger, 2023-04-29:
Accidentally I found a way to calibrate the magnetometer.
After decreasing "Starting Calibration Pulse" drastically (increasing the difference to "Stop Pulse"), the tracker now turns fast enough to perform an over-360°-turn in the 10 seconds, supposed to cover all angles. Now, a calibration run yields a 400° turn.

Guess what?

Absolute angle accuracy now is better than 3°. Thank you Raul and all others for the great design u360gts.

Please include this finding in your documentation as a solution. The problem took a lot of time, not to be supposed to happen any once more.

Settings now are:
Stop Pulse: 1502
Minimum Increment: 20
Max Gain: 700
Starting Calibration Pulse:
  pan calibration: Not less than 1460, otherwise calibration does not finish
  mag: 1270

PS: The dependency of both calibrations from the Starting Calibration Pulse possibly is the solution to my initial observation of mag calibration never succeeds versus pan calibration does it - somehow. Must have been a different "Starting Calibration Pulse" width.

AlexC176, 2023-10-27:
I ran into exactly the same issue. Starting calibration pulse needs to be smaller (higher rate) to complete the full 360o for mag calibration, but larger for correct pan calibration. I'm making compass code changes to detect a full tracker turn, versus the current 10 second (fix) mag calibration interval regardless of speed.

pki791, 2024-09-02:
@AlexC176 Have You finished the changes?

AlexC176, 2024-09-03:
I'm actively working on (and testing) the revised pan calibration code, it is not ready for a merge yet.

AaryanL, 2024-09-03:
> I'm actively working on (and testing) the revised pan calibration code, it is not ready for a merge yet. Thank you for asking.

Can you please post the line from the code with current 10 second (fix) mag calibration interval
I will increase it to 20, witch might help to calibrate it before your fix.

pki791, 2024-09-03:
It should be line 115 in src/main/sensors/compass.c
```
			if ((nextUpdateAt - tCal) < 10000000){//30000000) {    // 30s: you have 30s to turn the multi in all directions
```