raul-ortega/u360gts #127: Bluepill: unknown device after flashing via demonstrator

whoim2, 2023-05-31:
im use this instructions: https://github.com/raul-ortega/u360gts/blob/master/docs/configuration-loading-firmware.md
select amv-open360tracker_11.1.0_BLUEPILL.hex file, select global erase and jump to user program, flashing process and verify succesfuly.
Return BOOT0 jumper to 0 position, connec to windows7 PC.
Unknown usb device..
Try 2 working boards.
Help please.

whoim2, 2023-05-31:
thnx, im try tomorrow

raul-ortega, 2023-06-01:
> u360gts does not implement any usb driver. Yo must use a USB-TTL adapter
> ti conect to any serial port (usart/softserial). El mié., 31 may. 2023
> 10:32, whoim2 *@*.

PA9 PA10 is the correct way. You hace to flash de boar using the usb-ttl.
Once the board is flashed coonect the Mag and the display and start
configuring.

whoim2, 2023-06-01:
Can't connect without mag? he hasn't arrived yet.

whoim2, 2023-07-02:
im connect qmc mag and display to i2c (b10/b11) and usb-uart to a9/a10. Flash again
[image]
return boot0 jumper to 0 position, boot1 pos 0 too.

Power bluepill, cant connect from u360gts configurator - com port opened but no data, no save changes. Led on bluepill blinked about 0.5hz.

whoim2, 2023-07-02:
wtf?
im power bluepill to 3.3v from usb-uart and its work. If im power from usb (5v adapter > usb cable > bluepill usb port) its not work )

closed 2023-07-02 by whoim2