raul-ortega/u360gts #128: bluepill DFU mode

IcemanDZ, 2023-10-12:
I use the USB to TTL adpatper and follow the instuctions to flash the boatloader, but cant find other more info to flash the hex files to the board, there is no instrutions on the github, and since Lobato said its in the facebook group, but I searched it, there is no detailed instructions, can anyone help?

raul-ortega, 2023-10-12:
Spanish first, translated  to English below:

Hola Luis, creo estabas haciendo la misma consulta en el repositorio
github, mejor por aquí. El firmware debes cargarlo con un adaptador USB-TTL
conectado a los pines PA9 PA10, más un GND y un VIN. Además, tienes que
cambiar la posición de uno de los jumpers amarillos, y pulsar el botón
blanco cuando des la orden de flashear desde el configurador. Cuando
terminas de flashear, tienes que desconectar alimentación, colocar el
jumper en la posición original, y alimentar otra vez el USB-TTL. Es
importante colocar el mag porque la controladora intentará detectarlo, y
puede quedar indefinidamente esperando a encontrarlo dejando el led rojo
indicando su falta. Si habillitas el display y tampoco está presente,
también puedes tener problemas de comunicación desde el configurador. Se me
olvidaba comentar que también es posible flashear usando un ST-Link
conectado al puerto que lleva para ello, pero en lugar del configurador
debes usar una aplicación específica de STM. El motivo de no poder usar el
puerto USB que incorpora la bluepill se debe a que esta no incorpora un
chip para controlarlo, lo debe hacer el propio firmware que se cargue en la
bluepill previamente. Esto es un problema porque no deja espacio suficiente
para el firmware del tracker, además de que el firmware del tracker no
gestiona el puerto por limitación de espacio en memoria. Finalmente,
decirte que hay bluepills que no tienen la capacidad de memoria suficiente
para ser flashea das con el firmware del tracker, motivo por el cual podría
no funcionar aún teniendo conectado un mag.

English (Google Translator):

Hello Luis, I think you were asking the same query in the github
repository, better here.  You must load the firmware with a USB-TTL adapter
connected to the PA9 PA10 pins, plus a GND and a VIN.  Additionally, you
have to change the position of one of the yellow jumpers, and press the
white button when you give the flash command from the configurator.  When
you finish flashing, you have to disconnect the power, place the jumper in
the original position, and power the USB-TTL again.  It is important to
place the mag because the controller will try to detect it, and it may wait
indefinitely to find it, leaving the red LED indicating its absence.  If
you enable the display and it is not present either, you may also have
communication problems from the configurator.  I forgot to mention that it
is also possible to flash using an ST-Link connected to the port provided
for it, but instead of the configurator you must use a specific STM
application.  The reason for not being able to use the USB port that the
bluepill incorporates is because it does not incorporate a chip to control
it, this must be done by the firmware itself that is loaded into the
bluepill previously.  This is a problem because it does not leave enough
space for the tracker firmware, in addition to the fact that the tracker
firmware does not manage the port due to memory space limitation.  Finally,
tell you that there are bluepills that do not have enough memory capacity
to be flashed with the tracker firmware, which is why it may not work even
with a mag connected.

####

Use u360gts configurator to flash de board.

IcemanDZ, 2023-10-26:
looks like I still cant flash it, this is way so confusing, I can add you on facebook?

raul-ortega, 2023-10-26:
The way to flash the bluepill is a, let's say, standard procedure, it
involves putting the controller in boot mode and using the pins described
using a USB-TTL interface.  You could use the betaflight or inav
configurator to load the previously downloaded firmware on your computer
and, also, you can use the STM tool, I mean Flash Loader Demonstrator.
Another way to flash is using an ST-Link interface.  However, it could be a
driver problem on your computer, or even a faulty bluepill.  Sometimes it
may even be that your bluepill does not have enough memory.  Request to
join the u360gts group, other users who may have had the same problem could
help you solve it.