raul-ortega/u360gts #132: [FR] Tangle-free mode

whoim2, 2024-10-26:
I have a lot of wires including hdmi and ethernet. It is not realistic to run them through the rotary unit.
I think if the tracker would reach the specified limit when rotating, and then scroll 359 degrees in the opposite direction, that would solve my problem.
I'd also like a separate setting to turn off continuous rotation when losing signal, leave a customizable range of back and forth rotation, or turn it off altogether.

raul-ortega, 2024-10-26:
The name of this proyect is u360gts... 360 ... There already exists 180
proyects.

closed 2024-10-26 by raul-ortega

whoim2, 2024-10-26:
> already exists 180

All other projects are a far cry from yours, unfortunately. But thanks for the answer, I'll think of something, like a mechanical limiter and a servo with a spring fuse.... I can only work in one direction.

raul-ortega, 2024-10-26:
I apologize about my previous answer, this is a common problem within
Spanish users. But is more than probably that no work in that direction
will be done, european rules are too restrictive and current HD devices are
nowadays enough to fly short distances arround home. Have you thought in
using a HDMI wireless repeater? And sliprings designed for HDMI
transmission?

whoim2, 2024-10-26:
I'll give it some thought, thanks. Maybe if I give up on the glasses, it will be enough for me to wire the reserve rotary ethernet device