raul-ortega/u360gts #139: Use RSSI_display will result wrong display with battery

HuanHuanTuanZ, 2024-12-24:
[image]
[image]

HuanHuanTuanZ, 2024-12-30:
Device：STM32F103C8T6 Bluepill, 11.4.0

AlexC176, 2025-04-03:
@HuanHuanTuanZ I am making changes in the display module, for other reasons.

Please can you explain the problem another way. I do not understand the issue you have posted.

HuanHuanTuanZ, 2025-04-04:
> [@HuanHuanTuanZ](https://github.com/HuanHuanTuanZ) I am making changes in the display module, for other reasons.
> 
> Please can you explain the problem another way. I do not understand the issue you have posted.

Please refer to the image. When the RSSI DISPLAY is turned on, the BATTERY indicator on the first line gets covered by the progress bar.

AlexC176, 2025-04-08:
Fixed in my branch, the bar is RSSI2. Will request merge into main branch once I'm done with the other changes.

HuanHuanTuanZ, 2025-04-08:
> 在我的分支中，条形是 RSSI2。完成其他更改后，将请求合并到 main 分支。
Thank you for your contribution. It would be even better if you tried to fix the error of resetting to zero