raul-ortega/u360gts #146: Compatible with mainstream flight controllers

sunyanmeng963, 2025-08-05:
Using F1 or F3 flight controllers has lower costs, but these flight controllers have basically been phased out and are quite expensive. Can they be compatible with the current mainstream flight controllers, such as F405/AT32? This would be of great benefit to the development of U360GTS.

raul-ortega, 2025-08-06:
The u360gts project is evolving into u360gts-ng (next generation), a completely new ecosystem. This transition involves a move towards more advanced hardware platforms, which will enable significant improvements, the integration of new features, and better interconnectivity with the most popular radio control and communication systems in the community. Intensive work has been done on this new version, and the official launch is expected to be announced during the last quarter of this year. Therefore, it is most sensible to wait for the recommended hardware to be presented before making a purchasing decision.

sunyanmeng963, 2025-08-21:
great，I'm starting to look forward to it.